DennisOSRM/hms-mqtt-publisher #6: Auto-publish docker images for several platforms

DennisOSRM, 2023-11-04:
As mentioned first here: https://github.com/lumapu/ahoy/issues/1189#issuecomment-1793503475

InScene, 2023-11-04:
Ich konnte jetzt ein Image für folgende Plattformen erstellen:
linux/amd64, linux/arm64, linux/arm7v7
Jetzt läuft es auch auf dem Raspberry Pi.

DennisOSRM, 2023-11-04:
Dockerfile to be added in PR #9.

InScene, 2023-11-05:
I have created a docker action to solve this problem. It works, but currently it takes a long time to execute. For me it takes 1:20 hour. But I think it's good enough for the first version.

DennisOSRM, 2023-11-05:
Very cool, thanks so much! I'll create credentials and add them to the action tomorrow (if time permits).

InScene, 2023-11-05:
> Very cool, thanks so much! I'll create credentials and add them to the action tomorrow (if time permits).

Perfekt. I also added the mqtt port as a variable to the docker file.

closed 2023-11-05 by DennisOSRM

DennisOSRM, 2023-11-05:
Initial release has been pushed and the action is running: https://github.com/DennisOSRM/hms-mqtt-publisher/actions/runs/6761967696

The docker images should show up [here](https://hub.docker.com/repository/docker/dennisosrm/hms-mqtt-publisher/) on DockerHub in a little while.

DennisOSRM, 2023-11-06:
Publishing to DockerHub has succeeded. Thanks again for the great work!

InScene, 2023-11-07:
> Publishing to DockerHub has succeeded. Thanks again for the great work!

You are welcome. 
But the credit goes to you. You wrote the tool and reverse-engineered the data.

Please add a description to the Docker Hub repository. If you login to the website, you can edit it.

Here is an example Text:

---
This image implements a tool that fetches the current telemetry information from the HMS-XXXXW-T2 series of micro-inverters and publishes the data in JSON format to an MQTT broker.

The code of the application can be found here:
https://github.com/DennisOSRM/hms-mqtt-publisher

Set the following environment variables when executing:
```
INVERTER_HOST='ip-address hms-inverter'
MQTT_BROKER_HOST='ip-address mqtt broker'
MQTT_USERNAME='mqtt user'
MQTT_PASSWORD='mqtt pass'
MQTT_PORT='mqtt server port'
```

DennisOSRM, 2023-11-18:
descriptions are now updated