DennisOSRM/hms-mqtt-publisher #17: Home Assistant Addon

dominikandreas, 2023-11-05:
I would like to run this application directly within home assistant as an addon. I do have a little experience with writing addons and think it shouldn't be too complicated. Has someone already started with that? Does it make sense to have a separate repository for the home assistant addon or directly create a PR for this one?

DennisOSRM, 2023-11-05:
Sounds great! Happy to consider a PR for this repo.

dominikandreas, 2023-11-09:
Before we close this issue, let's adress these todos:

- update URLs in repository.json and config.yaml
- publish data with home assistant discovery support (add an option to the CLI for this?)
- get mqtt user, pw via bashio

I will create separate PRs for this

DennisOSRM, 2023-11-09:
Sounds good. Let's do separate PRs. Do you have a link with more information on the home assistant discovery support? That sounds interesting.

dominikandreas, 2023-11-09:
Sure, here's the documentation: 
https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery

And here's an example from Chatgpt for this codebase (haven't checked yet if it's actually sensible):
https://chat.openai.com/share/51baad75-6bac-4a52-bfa2-e02bf4733987

dominikandreas, 2023-11-09:
Another option in home assistant would be to directly use the API instead of going through the mqtt server: https://developers.home-assistant.io/docs/api/rest/
(I think the "POST /api/states/<entity_id>" endpoint should do the trick)

Most people have an mqtt server integrated into home assistant anyway, so not sure if it's necessary / any advantage to go through the API

DennisOSRM, 2023-11-10:
Oh, that's interesting too. It seems the HTTP request could be done using curl from inside the start script. What do you think is the better, or more canonical approach?

dominikandreas, 2023-11-10:
It would probably be better to do that within the rust application directly via a cli option, but I haven't done or seen that in an addon yet that makes use of the API that way. I don't see why it shouldn't be possible though or even be the preferred mode of operation. I'll look into it and do some testing

DennisOSRM, 2023-11-10:
Shouldn’t be too hard to implement it inside the tool itself. Let me know what you come up in your tests.

dominikandreas, 2023-11-11:
I tested publishing via API directly and it works (see [here](https://github.com/dominikandreas/hms-mqtt-publisher/compare/main...dominikandreas:hms-mqtt-publisher:home_assistant_rest_api) if you're curious), but it doesn't seem to be possible to group entities together in a single device. Having all entities as a single device in home assistant makes it easier to inspect the states and add all entities to a dashboard. 

I would therefore opt for the MQTT approach, seems to be the most common one anyway.

DennisOSRM, 2023-11-11:
Alright, then let's do it the MQTT way. Thanks for conducting the experiment. That's really helpful.

DennisOSRM, 2023-11-24:
PR #27 has been merged.

closed 2023-11-24 by DennisOSRM