DennisOSRM/hms-mqtt-publisher #22: Question: Power Adjustments

jogybaer0815, 2023-11-10:
Hi, just tested this library and it works great!
But I do have a question: Is there a possibility to configure the power ratio? I saw that you're working with a TCP connection to read values from the DTU and I was wondering if it's also possible to adjust the maxmimum generated power through this connection.
I was planning to handle this automatically but I'm currently struggling to set it up.

Thanks for your work!

DennisOSRM, 2023-11-10:
Yes, as far as I can tell, it should be possible to adjust the power ratio. It will be a matter of implementing the right call.

jogybaer0815, 2023-11-10:
Hi, thanks for the fast response. Is there any documentation that can be looked for those endpoints?

DennisOSRM, 2023-11-17:
I used the following writeup for coding: https://github.com/henkwiedig/Hoymiles-DTU-Proto/blob/main/protcol.md

Not sure how accessible it is, tho.

henkwiedig, 2023-12-13:
Setting the Powerlimit is shown here: https://github.com/henkwiedig/Hoymiles-DTU-Proto/blob/main/SetPowerLimit.py