DennisOSRM/hms-mqtt-publisher #28: No data from HMS

Wiki591, 2023-11-18:
Hi,
at first Thank You for sharing this project.

I've compiled the sources using RaspberryPi, Ubuntu 20.04 and cargo/rust 1.73.0 for a friend. Starting the binary with given ip-adresses of inverter and mqtt broker tells me, that the inverter is online. But nothing else happens, no data are published. The inverter is a HMS 800W-T2, checking the version in the hoymiles app it tells V1.1.15, I suppose its the version number of the DTU.

What I didn't do was setting the environment variables.

Any ideas?

T3mplate-wow, 2023-11-18:
May you have the same problem like I had. 
I thought that there are no data’s incoming, but there is just no auto discovery. That means no entity will be there automatically in HA or ioBroker or whatever. 
you have to fetch the data with a manual device.

Wiki591, 2023-11-18:
Erm, I am trying to build a bridge to domoticz using a small python script, subscribing and listening for topic hms800wt2/#

mosquitto_sub -t 'hms800wt2/#' -h [broker-ip] gives no data at all.

Wiki591, 2023-11-18:
Ähm, the version I posted is the version of the app which my friend sent to me, sorry. I asked him once again, inverter software version V1.00.08, HW H00.04.00, DTU software V00.01.08, HW H00.01.00

Wiki591, 2023-11-18:
OK, after some time of reading on discord I get more transparency in my issue. I tried to connect and read the data from the hms using the existing wlan, the inverter connected to a fritzbox. Did I learn right, that is not the way the tools is designed for?

Is there somewhere a howto which summarizes the prerequesites to collect the data from the inverter?

DennisOSRM, 2023-11-18:
If you get the message that the inverter is online, then we have received a valid response with the current readings from it. In other words, it's actually reachable on the network. This would point to the MQTT broker not being reached as expected.

If you set the `RUST_LOG=debug` environment variable, then you should get (much) more detailed logging.

Wiki591, 2023-11-18:
Do I have to set RUST_LOG during compilation or would it be sufficient to set it before starting the binary?

My work is a little bit complicate because my friend owns a raspi 3, with which compiling the tool is really no fun. So I am compiling here at home and carrying the binary with me on my next visit.

DennisOSRM, 2023-11-18:
The environment variable is read at runtime. There is no need to recompile, and setting it at start of the binary should be enough.

Wiki591, 2023-11-18:
I appreciate your support. Reporting any results will take some days.