DennisOSRM/hms-mqtt-publisher #33: Not receiving any updates

dominikandreas, 2023-11-22:
I'm not sure where this issue stems from, but it seems that either the inverter stops sending updates or the application gets stuck after a while (usually after a couple of hours). I'm still at the beginning of investigating why this happens. For now, I've increased the delay between updates to one minute to see if that fixes things.

DennisOSRM, 2023-11-23:
Oh, interesting. Curious if you get any insight into why this happened.

markus-at-git, 2023-11-27:
First of all, I wanted to thank you for the great work! But now unfortunately to the actual point: Since the update to version 0.2 I no longer receive any data. The inverter is displayed as "online", but no more data is sent via MQTT. I have restarted the add-on (Home Assistant) and the inverter, but unfortunately neither was successful. Reinstalling the add-on did not help either. Is there anything else I can do?

dominikandreas, 2023-11-27:
Sorry to hear that. Is there anything in the logs of the addon that would indicate what's going on?

markus-at-git, 2023-11-27:
Unfortunately not, everything actually looks good here.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Hoymiles HMS Wifi Addon
 Publishes telemetry information from Hoymiles HMS-XXXXW-T2 micro-inverters to your home assistant MQTT broker
-----------------------------------------------------------
 Add-on version: 0.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.1  (amd64 / qemux86-64)
 Home Assistant Core: 2023.11.3
 Home Assistant Supervisor: 2023.11.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[10:42:42] INFO: Configuring timezone (Europe/Berlin)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2023-11-27T10:42:43 [INFO] - inverter host: 192.168.178.175
2023-11-27T10:42:43 [INFO] - Inverter is Online
```

dominikandreas, 2023-11-27:
I assume that it somehow isn't able to communicate with your mqtt server. What does your config look like? Are you able to connect to the MQTT server via those settings? If the MQTT server is integrated in home assistant, try removing the settings apart from the inverter host to let the addon figure it out on its own via bashio.

markus-at-git, 2023-11-27:
In version 0.1, it was able to communicate with the MQTT server without any problems (the data has not changed either). I have now removed the settings from the configuration anyway, but without success. I can also log in to the MQTT Explorer with the same data without any problems.

dominikandreas, 2023-11-27:
I just realized that an update to the changelog somehow didn't make it in the last PR. It should've mentioned that the MQTT topics changed in favor of the home assistant discovery protocol which registers a device directly in home assistant.  

If you go to Settings -> Devices & services -> MQTT, you should see a device with the name "Hoymiles HMS-WiFi <first 8 digits of the serial number>" that includes all sensors in one place

markus-at-git, 2023-11-28:
Thanks for the tip. I have now looked in the "MQTT" integration, but only see a different element there, not Hoymiles. Do I understand correctly that the inverter should send its data to "homeassistant/..."? I saw data there once yesterday morning, but it was all "empty" and has not been updated since then. I then deleted the topic to make sure it wasn't old data and it hasn't come back since. Do I have to configure anything special in the MQTT integration? The broker is running and numerous other elements are sending their data there diligently (e.g. Tasmota, Shelly...)

markus-at-git, 2023-11-28:
I have now looked in the logs of the MQTT Broker and find the following entries there:

```
2023-11-28 10:56:11: New connection from 172.30.32.2:42044 on port 1883.
2023-11-28 10:56:11: Client <unknown> closed its connection.
2023-11-28 10:58:11: New connection from 172.30.32.2:35120 on port 1883.
2023-11-28 10:58:11: Client <unknown> closed its connection.
2023-11-28 11:00:11: New connection from 172.30.32.2:39592 on port 1883.
2023-11-28 11:00:11: Client <unknown> closed its connection.
2023-11-28 11:02:11: New connection from 172.30.32.2:35902 on port 1883.
2023-11-28 11:02:11: Client <unknown> closed its connection.
```

They are only written when I have the Hoymiles HMS Wifi Addon running and repeat every 2 minutes. So I assume that there is actually a problem with the connection to the MQTT server. Just as you already suspected. :D

Now the question is where exactly the problem lies. As I said, so far there have been no problems with the connection and it also works with all other devices without any problems. Unfortunately, the logs from the Hoymiles HMS Wifi Addon do not provide any information. Can I get more information from somewhere else?

dominikandreas, 2023-11-28:
Adding some debug logging is not super difficult todo, I will try to push something later today

Seb0815, 2023-11-29:
I have the exact same issue (not receiving any data with the latest build). 
I'm using the image from dockerhub and "pass" inverter_host/mqtt_broker_host/mqtt_port via environment variable to the service (just use the simple mqtt broker without homeassistant). 
With v0.1 all is working, with v0.2 I only get the message that inverter is online/offline (via container log) but nothing else is send via MQTT. I also noticed that with v0.2 no communication between the inverter and Hoymiles Cloud/App did happen. If I switch the version back to v0.1 I got data via mqtt AND Hoymiles Cloud immediately.

Any way to debug/troubleshoot the MQTT communication with inverter?

dominikandreas, 2023-11-29:
I pushed some changes to my PR: https://github.com/dominikandreas/hms-mqtt-publisher 
The nightly addon should show debug logs when enabled via the config.

Note that MQTT is only used to communicate with home assistant, the communication with the inverter should be the same as before.

I hope the debug logs shed some more light on the issues

DennisOSRM, 2023-11-29:
Btw, you can set the log level without code changes by an environment variable. Eg

```
$ RUST_LOG=debug ./main
```

DennisOSRM, 2023-11-29:
Thinking a bit more about that, one thing that needs to be changed is to make the `publish(.)` call to MQTT non-blocking, i.e. use `.try_publish(.)`. I couldn't find proof by searching on the internet, yet it seems that this may cause the hanging behaviour.

dominikandreas, 2023-11-29:
That's exactly what I did. I should

> Thinking a bit more about that, one thing that needs to be changed is to make the `publish(.)` call to MQTT non-blocking, i.e. use `.try_publish(.)`. 

That's exactly what I did. I'll open a PR, just haven't had time to clean it up

DennisOSRM, 2023-11-29:
It's also part of PR #42 that I plan to merge tomorrow once the sun is up.

DennisOSRM, 2023-11-30:
One brief update to share: i was able to reproduce the issue in my setup. Some or all values were either not updated or sending messages was stuck at all. I am running a potential fix in my test setup already that will use non-blocking try_publish message, as well as trying to repeat a number of times in case of error.

DennisOSRM, 2023-11-30:
Commit https://github.com/DennisOSRM/hms-mqtt-publisher/commit/b233e5429a86472dfbbdd979e64d2f7c888f7a62 in main branch should fix the issue

DennisOSRM, 2023-12-02:
The issue seems to be fixed. E.g. I am getting reliable measurements since the sun came up.

[image]

DennisOSRM, 2023-12-03:
Interestingly, I had just observed a case of values not being updated. Will investigate.

ckmde, 2023-12-05:
> I'm not sure where this issue stems from, but it seems that either the inverter stops sending updates or the application gets stuck after a while (usually after a couple of hours). I'm still at the [...]

I've found the same problem here. I've added socket read and write timeouts. Because rust is not using anyone by default and so if the Inverter is not answering, the application gets stuck. Had this a few times. After this changes no more locks.

inverter.rs

        stream.set_write_timeout(Some(Duration::new(45, 0))).expect("set_write_timeout call failed");

        stream.set_read_timeout(Some(Duration::new(45, 0))).expect("set_read_timeout call failed");

DennisOSRM, 2023-12-05:
Thanks for the update. Will incorporate this change.

DennisOSRM, 2023-12-06:
Implemented in https://github.com/DennisOSRM/hms-mqtt-publisher/pull/57

DennisOSRM, 2023-12-10:
A new release was just published including docker images. This includes all the most recent bug fixes and should fix the update issues. Please give this a try.

closed 2023-12-10 by DennisOSRM