DennisOSRM/hms-mqtt-publisher #45: Docker v.02 broken

InScene, 2023-11-30:
Hi,
I have updated the docker image from v.01 to v.02. Now it is no longer working and i get following errors:

`toml config unparsable: Error { inner: Error { inner: TomlError { message: "invalid string\nexpected `\"`, `'`", original: Some("inverter_host = \"\" \n\n[home_assistent] \nhost = \"\"\nusername = \"\"\npassword = \"\"\nport = \n\n"), keys: [], span: Some(84..85) } } }
thread 'main' panicked at src/main.rs:46:52: `

I changed back to v.01 and it worked again. Are there any new settings required or is it a bug?

w0d4, 2023-11-30:
I get the exact same error message since updating docker image.
Here is the full trace with `RUST_BACKTRACE=full` in case it helps.

```
thread 'main' panicked at src/main.rs:46:52:
toml config unparsable: Error { inner: Error { inner: TomlError { message: "invalid string\nexpected `\"`, `'`", original: Some("inverter_host = \"\" \n\n[home_assistent] \nhost = \"\"\nusername = \"\"\npassword = \"\"\nport = \n\n"), keys: [], span: Some(84..85) } } }
stack backtrace:
 0:     0x55bbda3e5b8c - std::backtrace_rs::backtrace::libunwind::trace::h67a838aed1f4d6ec
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
 1:     0x55bbda3e5b8c - std::backtrace_rs::backtrace::trace_unsynchronized::h1d1786bb1962baf8
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
 2:     0x55bbda3e5b8c - std::sys_common::backtrace::_print_fmt::h5a0b1f807a002d23
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/sys_common/backtrace.rs:67:5
 3:     0x55bbda3e5b8c - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hf84ab6ad0b91784c
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/sys_common/backtrace.rs:44:22
 4:     0x55bbda40fb5c - core::fmt::rt::Argument::fmt::h28f463bd1fdabed5
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/fmt/rt.rs:138:9
 5:     0x55bbda40fb5c - core::fmt::write::ha37c23b175e921b3
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/fmt/mod.rs:1114:21
 6:     0x55bbda3e282e - std::io::Write::write_fmt::haa1b000741bcbbe1
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/io/mod.rs:1763:15
 7:     0x55bbda3e5974 - std::sys_common::backtrace::_print::h1ff1030b04dfb157
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/sys_common/backtrace.rs:47:5
 8:     0x55bbda3e5974 - std::sys_common::backtrace::print::hb982056c6f29541c
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/sys_common/backtrace.rs:34:9
 9:     0x55bbda3e6fe3 - std::panicking::default_hook::{{closure}}::h11f92f82c62fbd68
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:272:22
10:     0x55bbda3e6d04 - std::panicking::default_hook::hb8810fe276772c66
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:292:9
11:     0x55bbda3e7565 - std::panicking::rust_panic_with_hook::hd2f0efd2fec86cb0
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:731:13
12:     0x55bbda3e7461 - std::panicking::begin_panic_handler::{{closure}}::h3651b7fc4f61d784
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:609:13
13:     0x55bbda3e60b6 - std::sys_common::backtrace::__rust_end_short_backtrace::hbc468e4b98c7ae04
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/sys_common/backtrace.rs:170:18
14:     0x55bbda3e71b2 - rust_begin_unwind
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:597:5
15:     0x55bbda1fcba5 - core::panicking::panic_fmt::h979245e2fdb2fabd
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/panicking.rs:72:14
16:     0x55bbda1fd083 - core::result::unwrap_failed::h8c4b86241881fbbb
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/result.rs:1652:5
17:     0x55bbda211451 - hms_mqtt_publish::main::h2c0396ca00af51ec
18:     0x55bbda20f353 - std::sys_common::backtrace::__rust_begin_short_backtrace::h125a104efa832901
19:     0x55bbda236af9 - std::rt::lang_start::{{closure}}::he9f8bc4f9782d444
20:     0x55bbda3ddb7b - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::hf9057cfaeeb252e2
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/ops/function.rs:284:13
21:     0x55bbda3ddb7b - std::panicking::try::do_call::h629e203a624883e4
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:504:40
22:     0x55bbda3ddb7b - std::panicking::try::h7b61614724d6a4f1
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:468:19
23:     0x55bbda3ddb7b - std::panic::catch_unwind::h354ac1c0268491d8
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panic.rs:142:14
24:     0x55bbda3ddb7b - std::rt::lang_start_internal::{{closure}}::h919fee3c5ba8f617
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/rt.rs:148:48
25:     0x55bbda3ddb7b - std::panicking::try::do_call::h54583f67455bff32
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:504:40
26:     0x55bbda3ddb7b - std::panicking::try::hb0e12c4e01d39dc2
[... cut ...]
27:     0x55bbda3ddb7b - std::panic::catch_unwind::h367b6339e3ca9a3b
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panic.rs:142:14
28:     0x55bbda3ddb7b - std::rt::lang_start_internal::ha5ce8533eaa0fda8
                             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/rt.rs:148:20
29:     0x55bbda211bb5 - main
30:     0x7f3d4a528d0a - __libc_start_main
31:     0x55bbda1fd38a - _start
32:                0x0 - <unknown>
```

dominikandreas, 2023-11-30:
It looks to me like the parameters in the config.toml are not provided. Before you could just pass them via command line arguments, now you need to pass them as environment variables during the build process, or create/edit the config.toml and mount it in the container.

I think `docker run --rm -it -v ./config.toml:/config.toml <docker image>` should work

w0d4, 2023-11-30:
Thanks. Container is working again.
I tried already something similar, but got the wrong remote path
Will see if data is transmitted as soon my inverter awakes tomorrow after sunrise.

DennisOSRM, 2023-12-01:
The main output to look for is 'The inverter is online'. We habe recently fixed a few issues with values not always getting updated.

w0d4, 2023-12-02:
Yes, everything is working with config file volume mappings.

DennisOSRM, 2023-12-02:
Cool, thanks for the feedback. Closing here.

closed 2023-12-02 by DennisOSRM

KFU141, 2023-12-08:
Hi @w0d4 for me it does not work, see output:
`docker run --rm -it -v ./config.toml:/config.toml dennisosrm/hms-mqtt-publisher:latest                                                      docker: Error response from daemon: create ./config.toml: "./config.toml" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.
`
Can you tell what I do wrong?