DennisOSRM/hms-mqtt-publisher #50: Add a tool to set the wifi configuration of the inverter

DennisOSRM, 2023-12-01:
Should be a matter of retrieving a `GetConfig` response, encode the changed information and play it back in a `SetConfig` request

dicer, 2023-12-15:
Since I want to run my inverter in a network without internet access and would also prefer to get it there without downloading any apps or register an account, it would be very helpful to have this tool available. Let me know if I can help test with my inverters

DennisOSRM, 2023-12-16:
That's a very good reason to have this feature. Implementing this will likely start early next year.

dicer, 2024-01-02:
I did a quick test last year and I can connect to the inverter just fine. I used the following script to read the config: https://github.com/henkwiedig/Hoymiles-DTU-Proto/blob/main/GetConfig.py

Don't have the config with me, but I'll upload it tonight. Maybe someone with an already provisioned inverter could paste their config here (with obfuscated *passwords*, wifi names,  and serial numbers of course). Just let me know if the wifi password is somehow encoded or just plain text like the default password.
Then I'll diff it and would try to send it to one of my inverters for testing.

dicer, 2024-01-09:
These are the fields I get from the above GetConfig run:
`time: 12627 netmode_select: 1 server_send_time: 15 serverport: 10081 meter_kind: 2.1.21_hm meter_interface: 2.1.21_hm wifi_ssid: DTU-W600 wifi_passward: 12345678 server_domain_name: dataeu.hoymiles.com dtu_sn: 111111111111 access_model: 1 cable_dns_0: 223 cable_dns_1: 6 cable_dns_2: 6 cable_dns_3: 6 wifi_mac_0: 11 wifi_mac_1: 111 wifi_mac_2: 111 wifi_mac_3: 11 wifi_mac_4: 111 wifi_mac_5: 11 dtu_ap_ssid: DTUBI-11111111 dtu_ap_pass: xxxxxxxx`

The following fields were obfuscated: dtu_sn, wifi_mac_x, dtu_ap_ssid and dtu_ap_pass

Could someone with a inverter connected to a wifi run the GetConfig script and post the values here please? With obfuscation of the mentioned fields plus wifi_ssid and wifi_passward.

That would be very helpful!

DennisOSRM, 2024-01-10:
Cool! I'll see if I can get it to run end of week.

DennisOSRM, 2024-01-11:
It's all there in clear text:

```
offset: 3600
time: 1704981867
netmode_select: 1
server_send_time: 15
wifi_rssi: 70
serverport: 10081
meter_kind: 2.1.21_hm 
meter_interface: 2.1.21_hm 
wifi_ssid: <redacted cleartext>
wifi_passward: <redacted cleartext>
server_domain_name: dataeu.hoymiles.com 
dtu_sn: <redacted sn>
access_model: 1
cable_dns_0: 223
cable_dns_1: 5
cable_dns_2: 5
cable_dns_3: 5
wifi_ip_addr_0: 192
wifi_ip_addr_1: 168
wifi_ip_addr_2: 4
wifi_ip_addr_3: 182
wifi_mac_0: <redacted cleartext>
wifi_mac_1: <redacted cleartext>
wifi_mac_2: <redacted cleartext>
wifi_mac_3: <redacted cleartext>
wifi_mac_4: <redacted cleartext>
wifi_mac_5: <redacted cleartext>
dtu_ap_ssid: <redacted cleartext>
dtu_ap_pass: <redacted cleartext>
```

DennisOSRM, 2024-01-11:
I have looked further into this issue and it seems this is pretty easy to achieve, i.e. will be adding a tool that is just setting wifi SSID and password, and dns servers.

dicer, 2024-01-11:
I was actually also wondering whether I need to send the whole config 
back (like in the SetConfig example in the other repo) or I could just 
send some of the values.

I could try to just set the wifi AP settings tomorrow. I'll let you know 
what happens ;)

serpivolant, 2024-01-18:
@DennisOSRM 
I have implemented this functionality in:

https://github.com/suaveolent/hoymiles-wifi

if you need any help with it.

Trying to send the wifi configuration only did, however, not work in my attempts. The only solutions seems to be to send the entire config back. The sMiles installer application also does this the same way.

suaveolent, 2024-01-18:
@DennisOSRM
I have implemented this functionality in:

https://github.com/suaveolent/hoymiles-wifi

if you need any help with it.

Trying to send the wifi configuration only did, however, not work in my attempts. The only solutions seems to be to send the entire config back. The sMiles installer application also does this the same way.

DennisOSRM, 2024-01-18:
Cool. Thanks so much for sharing. This is super helpful.

dicer, 2024-01-25:
I tried setting the wifi with the hoymiles-wifi script by @suaveolent and 
can confirm that it works with an inverter straight out of the box! I 
have some connection problems, but that might be a local problem and I 
have to debug further.
Can't reach the inverter when it is in the target wifi and it seems to 
drop the connection frequently. Might be because of wifi strength or 
because it can't reach the internet through that wifi.

I'll report back soonish

suaveolent, 2024-01-25:
> I tried setting the wifi with the hoymiles-wifi script by @suaveolent and can confirm that it works with an inverter straight out of the box! I have some connection problems, but that might be a [...]

Thanks for letting me know. Might also be due to an old firmware. 
I will look into how firmware updates are supplied and maybe add the functionally to the library.

dicer, 2024-01-25:
@suaveolent Maybe have a look here: https://github.com/veganmosfet/Balcony_in_the_cloud#firmware-update
@veganmosfet might be able to help. He triggered it through the cloud connection though, but there possibly is a more direct way to trigger the firmware download/update.
It won't be as straight forward as setting the wifi info though, cause it likely involves some DNS bending and supplying the firmware image locally (see `Post-patch reproduction` in the above repo)

I also though about just changing the `server_domain_name` config property and point it back to the IP/host where the script is running. You could open a webserver, accept the connection from the DTU and request it to download a firmware update.

veganmosfet, 2024-01-26:
I don't know for the HMS devices but for the HM device you can just send an update command locally from a PC to the DTU and then the DTU will fetch the firmware. 
I involved the cloud, so only part of the url was under control. However, the command from the cloud to the DTU includes the full url, so no need to mess up with dns.
My proposal: you trigger an update from the cloud and at the same time record the Ethernet traffic of your inverter. Then you can extract the local "update" command, the full url is included in plaintext there in the TCP packet content.

suaveolent, 2024-01-27:
> I don't know for the HMS devices but for the HM device you can just send an update command locally from a PC to the DTU and then the DTU will fetch the firmware. I involved the cloud, so only part [...]

Sounds like a good idea. Will try to capture the relevant packages. 
I will also try to capture the relevant packages to set the grid profile.

suaveolent, 2024-01-29:
Making some progress:
The firmware is loaded from:

http://fwupdate.hoymiles.com/cfs/bin/2311/06/1488725943932555264.bin

suaveolent, 2024-01-29:
@dicer I have released a new version (v.0.0.8) which should implement the firmware upgrade, if you feel lucky.
I cannot confirm if it works, since I have already installed the V00.01.11 before.
However, when issuing the command, the inverter goes offline for a couple of minutes and then resumes operation. 

I will monitor the network in the future, to see if it really executes the command.

suaveolent, 2024-01-30:
@dicer After inspecting the network traffic I can confirm: after issuing the firmware update the inverter downloads the firmware from the hoymiles server. 

This means we probably can also host the firmware somewhere else.

dicer, 2024-02-03:
I moved the discussion about updating the firmware over to https://github.com/suaveolent/hoymiles-wifi/pull/1
No need to spam this issue any more, which is about setting the wifi config (which works for me, but the inverter keeps dropping from the wifi very regularly).

But maybe someone is able/willing to try connecting their unregistered(!) inverter to the internet (with the tool by @suaveolent)  and see if the inverter behaves the same if the cloud is reachable. I'm running the inverter in a wifi with no internet connection and am still unsure if the wifi dropping is related to not being able to connect to the internet at all, not being registered with the cloud, the old firmware or some local problem.

DennisOSRM, 2024-02-03:
I would assume the wifi issue to be unrelated to cloud connectivity. My inverter is isolated from the internet, i.e. all outgoing packages from it are dropped at the gateway and the connection to the wifi is pretty solid.

dicer, 2024-02-03:
I see exactly 94 pings to the inverter succeed, then around 230-250 pings fail and then it comes back for exactly 94 pings again and so on. So for me it pretty solidly fails ;)

teclab-at, 2024-04-26:
> I see exactly 94 pings to the inverter succeed, then around 230-250 pings fail and then it comes back for exactly 94 pings again and so on. So for me it pretty solidly fails ;)

Same issue with me. The HMS-800W-2T establishes connection every 15 minutes for a few seconds and drops its data to the cloud. The other time the WiFi seems of.