DennisOSRM/hms-mqtt-publisher #56: Docker image configuration is not working as described

thielepaul, 2023-12-04:
I just tested the docker image and the configuration through env variables is not working yet.

Looking at your dockerfile (https://github.com/DennisOSRM/hms-mqtt-publisher/blob/main/Dockerfile#L38) it looks like this line should be part of an entrypoint script instead of being executed during build with `RUN`.

I can also create a PR if you are interested.

```
$ docker run --rm -it dennisosrm/hms-mqtt-publisher cat config.toml

inverter_host = ""

[home_assistent]
host = ""
username = ""
password = ""
port =
```

AFPass, 2023-12-05:
I can't start Docker either.

What do i have to do?
Thanks

`thread 'main' panicked at src/main.rs:46:52:
toml config unparsable: Error { inner: Error { inner: TomlError { message: "invalid string\nexpected `\"`, `'`", original: Some("inverter_host = \"\" \n\n[home_assistent] \nhost = \"\"\nusername = \"\"\npassword = \"\"\nport = \n\n"), keys: [], span: Some(84..85) } } }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

dominikandreas, 2023-12-05:
Right, the environment variables used to build the container are only used during the build step and can't be passed later on. 

Having an entrypoint script that essentially does the same thing as the line in the Dockerfile that creates the `config.toml` should work fine. Feel free to open a PR for this.

To get it running in the meantime, you can create the `config.toml` locally and the run the container via `docker run --rm -it -v ./config.toml:/config.toml dennisosrm/hms-mqtt-publisher`

DennisOSRM, 2023-12-05:
I will think of a better way to supply the configuration at runtime. Seems this is bothering too many people at this point.

DennisOSRM, 2023-12-06:
I have a first draft in https://github.com/DennisOSRM/hms-mqtt-publisher/pull/61. It will simply override anything that's in the toml file (or not) at startup when it is defined in environment variables. Not super elegant, but should get the job done.

DennisOSRM, 2023-12-06:
Oh, that one is interesting. Thanks for sharing.

marchorst, 2023-12-11:
Why not simply use a Docker Compose file and assign the configuration file?

`version: '3'

services:
  hms-mqtt-publisher:
    image: dennisosrm/hms-mqtt-publisher
    volumes:
      - ./config.toml:/config.toml
   `

iSnackyCracky, 2023-12-15:
It looks like there's also a typo in this config :D

It should be `[home_assistant]` instead of `[home_assistent]`
I almost pulled my hair out trying to figure out why nothing was happening after providing the config.toml as a volume to the container and the log just seemed to stop after "Inverter is Online" (even with `RUST_LOG=debug`)

After changing to `[home_assistant]` and restarting the container once more, messages were instantly published via MQTT and received in Home Assistant!

DennisOSRM, 2023-12-19:
Typo has been fixed in the latest main branch. Thanks again for reporting.

FRWDBurg, 2024-03-02:
Hello,

I have the same problem:
`thread 'main' panicked at src/main.rs:55:52:
toml config unparsable: Error { inner: Error { inner: TomlError { message: "invalid string\nexpected `\"`, `'`", original: Some("inverter_host = \"\" \n\n[home_assistant] \nhost = \"\"\nusername = \"\"\npassword = \"\"\nport = \n\n"), keys: [], span: Some(84..85) } } }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`
Try to start the hms-publisher using the following docker compose file:

    hmsMQTTPublisher:
      image: dennisosrm/hms-mqtt-publisher:latest
      container_name: hmsMQTTPublisher
      environment:
        INVERTER_HOST: 192.168.178.59
        MQTT_BROKER_HOST: 192.168.178.40
        MQTT_USERNAME: 
        MQTT_PASSWORD: 
        MQTT_PORT: 1883
      restart: unless-stopped

Can someone help me and tell me where the problem is?
Thank you

FRWDBurg, 2024-03-06:
Sorry for asking, but I can't get any further with tsfuchs' advice.
I created the config.toml and tried to start it with compose.yaml. There is the following error:
`
Failure
failed to deploy a stack: Network hms_mqtt_publisher_default Creating Network hms_mqtt_publisher_default Created Container hms_mqtt_publisher-hms-mqtt-publisher-1 Creating Container hms_mqtt_publisher-hms-mqtt-publisher-1 Created Container hms_mqtt_publisher-hms-mqtt-publisher-1 Starting Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/data/compose/8/config.toml" to rootfs at "/config.toml": mount /data/compose/8/config.toml:/config.toml (via /proc/self/fd/6), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
`

thielepaul, 2024-03-06:
> Sorry for asking, but I can't get any further with tsfuchs' advice.
> I created the config.toml and tried to start it with compose.yaml. There is the following error:
> `
> Failure
> failed to deploy a [...]

This sounds like the config.toml file is either not existing or not at the correct path

FRWDBurg, 2024-03-09:
Hello @thielepaul, thank you very much for your support. It's almost embarrassing, but the config.toml was in a config.toml folder.
I didn't recognize via the console that it was a folder and not a file. Everything is working now.

PR #70 (2025-01-05): Fix/docker read environment vars
- [x] run `cargo clippy` and fix all issues
- [x] run `cargo fmt` to format all source files

Fixes #56
closed 2025-01-05 by DennisOSRM