DennisOSRM/hms-mqtt-publisher #88: Home Assistant addon fixed on version 0.3

ordo01, 2024-01-29:
Hi,
in the dockerfile for the home assistant addon is still fixed to version 0.3. 
Can you please correct this to make the latest release available as home assistant addon as well. 
Thanks

ordo01, 2024-01-29:
And please check the dockerfile for the nightly build addon as well. This is pointing to the fork from dominikandreas and not your GitHub repository

dominikandreas, 2024-01-29:
Hey thanks for the reminder, I'll likely have time again to work on this end of the week (a bit swamped with other responsibilities at the moment unfortunately).

dominikandreas, 2024-01-29:
Note that the nightly addon is really just meant for nightly testing before upgrading the main one. Since I was the one testing things, I left it pointing to my repo to play around with things independently and also because the corresponding GitHub action workflow didn't exist back then. I guess we can / should change this now