DennisOSRM/hms-mqtt-publisher #104: hms800w-2t stops accepting connections randomly

Marv1995, 2024-03-03:
hello,
my hms800w-2t stops acception connections after some days.
it will work perfectly for 3-4 days then it stops.
someone has a fix for that ?

you can even restart the panel via the app, nothing happens.
plugging it out of the wall and back it, nothing happens.
the next day everything is fine again.

config.toml:
inverter_host = "<INVERTER IP>"
coop_mode = true
update_interval = 60500

[simple_mqtt]
host = "<MQTT IP>"
port = 1883

Logs:
`[2024-03-03T08:43:11Z DEBUG hms2mqtt::simple_mqtt] dtu_sn: <censored> time: 1709455397 device_nub: 1 package_nub: 1 inverter_state {inv_id: <censored> port_id: 1 grid_voltage: 2369 grid_freq: 5005 pv_current_power: 2369 unknown1: 99 unknown2: 999 temperature: 215 unknown3: 5 unknown4: 1 bit_field: 7143442 11: 59656} port_state {pv_sn: <censored> pv_port: 1 pv_vol: 329 pv_cur: 758 pv_power: 2502 pv_energy_total: 29064 pv_daily_yield: 254 bitfield: 50921472} port_state {pv_sn: <censored> pv_port: 2 pv_vol: 4 pv_cur: 1 bitfield: 50331648} pv_current_power: 2369 pv_daily_yield: 254
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/inverter_local_time, Pkid = 0, Payload Size = 29
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_current_power, Pkid = 0, Payload Size = 5
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_daily_yield, Pkid = 0, Payload Size = 3
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_current_power, Pkid = 0, Payload Size = 5
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_daily_yield, Pkid = 0, Payload Size = 3
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_grid_voltage, Pkid = 0, Payload Size = 5
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_grid_freq, Pkid = 0, Payload Size = 5
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_inv_temperature, Pkid = 0, Payload Size = 4
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port1_voltage, Pkid = 0, Payload Size = 4
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port1_curr, Pkid = 0, Payload Size = 4
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port1_power, Pkid = 0, Payload Size = 5
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port1_energy, Pkid = 0, Payload Size = 5
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port1_daily_yield, Pkid = 0, Payload Size = 3
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port2_voltage, Pkid = 0, Payload Size = 3
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port2_curr, Pkid = 0, Payload Size = 4
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port2_power, Pkid = 0, Payload Size = 1
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port2_energy, Pkid = 0, Payload Size = 1
[2024-03-03T08:43:11Z DEBUG rumqttc::state] Publish. Topic = hms800wt2/pv_port2_daily_yield, Pkid = 0, Payload Size = 1
[2024-03-03T08:43:13Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 2631 millisecs
[2024-03-03T08:43:18Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:43:23Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:43:28Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5001 millisecs
[2024-03-03T08:43:33Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:43:38Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5001 millisecs
[2024-03-03T08:43:43Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:43:48Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:43:53Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 4999 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:43:58Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5000 millisecs
[2024-03-03T08:44:03Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5002 millisecs,
                last outgoing request before 5002 millisecs
[2024-03-03T08:44:08Z DEBUG rumqttc::state] Pingreq,
                last incoming packet before 5000 millisecs,
                last outgoing request before 5001 millisecs
[2024-03-03T08:44:11Z DEBUG hms2mqtt::inverter] could not connect: Connection refused (os error 111)
[2024-03-03T08:44:11Z INFO  hms2mqtt::inverter] Inverter is Offline`

DennisOSRM, 2024-03-03:
Interesting. Could it be a weak wifi signal?

Marv1995, 2024-03-03:
No, its literally 2m away, just a window inbetween ^^

Marv1995, 2024-03-03:
Just tested again, unplugged the whole system, waited 5 min, plugged back in, waited another 5 min, started the mqtt publisher, connection refused. second test, send a restart signal via app, waited 5 min, again connection refused. its pretty weird. Tomorrow everything will be okay again, i dont know why ^^

suaveolent, 2024-03-05:
@Marv1995 @DennisOSRM this might be related to:

https://github.com/suaveolent/ha-hoymiles-wifi/issues/14

Still assessing the underlying problem.

DennisOSRM, 2024-03-09:
Any further insights into the issue, @suaveolent ?

Marv1995, 2024-03-09:
just an update, without touching the device, rejection connections randomly even throughout the day (inverter -> publisher (connection refused) -> nodered -> influx -> grafana, blue graph):
[image]

suaveolent, 2024-03-10:
> Any further insights into the issue, @suaveolent ?

Not yet, but restarting the DTU via the restart command (either via Cloud or locally via e.g. my custom component) seems to remove the update limit.

Another idea: Maybe it is an issue with the Hoymiles cloud? Since the issue appeared independent of the used integration.

Marv1995, 2024-03-17:
quick update, after updating my firmware from V00.01.05 to V00.01.11 the issue seems to be solved (7 days of testing now). if it happens again i will provide you with another update.

Marv1995, 2024-04-02:
no problem since the firmware update. we can close this issue.