DennisOSRM/hms-mqtt-publisher #106: env_logger didn't have the feature anstream

Hobo2k, 2024-03-08:
Hi,

I can't compile it and don't find an solution:

```
Updating crates.io index
   error: failed to select a version for `env_logger`.
   ... required by package `hms-mqtt-publish v0.2.0 (/root/hms-whg1)`
versions that meet the requirements `^0.11.2` are: 0.11.3, 0.11.2

the package `hms-mqtt-publish` depends on `env_logger`, with features: `anstream` but `env_logger` does not have these features.
It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.

failed to select a version for `env_logger` which could resolve this conflict
```
 

Thanks for your help.

DennisOSRM, 2024-03-10:
if you are not using latest main branch, use `git pull` to fast forward. Then try issuing the following three command:

```
$ cargo clean
$ cargo update
$ cargo build
```

That _should_ help.

Hobo2k, 2024-03-10:
Didn't work
`cargo clean
root@hms-w-2t:~/hms-whg1# cargo update
    Updating crates.io index
error: failed to select a version for `env_logger`.
    ... required by package `hms-mqtt-publish v0.2.0 (/root/hms-whg1)`
versions that meet the requirements `^0.11.2` are: 0.11.3, 0.11.2
the package `hms-mqtt-publish` depends on `env_logger`, with features: `anstream`
but `env_logger` does not have these features.
 It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.
failed to select a version for `env_logger` which could resolve this conflict
root@hms-w-2t:~/hms-whg1# cargo build
    Updating crates.io index
error: failed to select a version for `env_logger`.
    ... required by package `hms-mqtt-publish v0.2.0 (/root/hms-whg1)`
versions that meet the requirements `^0.11.2` are: 0.11.3, 0.11.2
the package `hms-mqtt-publish` depends on `env_logger`, with features: `anstream`
but `env_logger` does not have these features.
 It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.
failed to select a version for `env_logger` which could resolve this conflict`

Hobo2k, 2024-03-19:
okay fixed it by updating to rustc 1.7x - in debian 1.63 is stable and seems to be too old

closed 2024-03-19 by Hobo2k