DennisOSRM/hms-mqtt-publisher #111: naming for multiple inverter

Hobo2k, 2024-03-19:
Hi,

I have two of the HMS-800W-2T and want's to request both inverters.
I tried it with configuring the normal and the nightly version. Is there another possibility?

The other thing is, it overwrites the other in HA, because of the serial is not complete in the naming:
[image]

Where I can edit it, that I have only the last 6 for example of the serial?

Thanks

KFU141, 2024-03-28:
Hi @Hobo2k,

I found a way doing that. You have to build docker image or simple run the publisher with rust after you adjusted the hard coded MQTT topics in the simple_mqtt.rs. 

On my end I now have two docker containers running with different MQTT topics which do not overwrite each other anymore.

I would also appreciate a build in solution which can handle multiple inverters in parallel.

Cheers

mpxxl, 2024-03-30:
I also have two docker and mqtt instances with different ports running.

dc7kr, 2024-04-08:
you can use my PR for that - I had the same issue and fixed the naming by using suffixes. Unfortunately still not merged due to concerns regarding breaking changes....

[https://github.com/DennisOSRM/hms-mqtt-publisher/pull/94]

DennisOSRM, 2026-10-04:
Thanks for reporting this! It's fixed by #150: run one instance per inverter and give each its own `DEVICE_ID` (or `device_id` in config.toml), for example the last digits of the serial number. The Home Assistant entities are then named after that id instead of the first 8 serial digits, which are the same for inverters from the same production week, and each instance gets its own MQTT client id. The nightly Docker image and the nightly add-on (option `device_id`) have it already; the stable ones follow with the next release.

closed 2026-10-04 by DennisOSRM