DennisOSRM/hms-mqtt-publisher #112: Graph is very pointy

simonszu, 2024-03-23:
I am using this project as anyone else, but am also exporting Home Assistant data via Prometheus and graphing it via Grafana. It seems that the grafana graph is quite pointy with lots of spikes. 
[image]

I am not sure if this is usual for balcony solar devices, since the sun isn't rapidly switching on and off, so i am opening this issue to confirm if this is the usual behaviour for the metric, or if there is something to be improved.

dominikandreas, 2024-03-23:
The spikes are typical for a day with a mix of sun and clouds. For straight and continuous sunshine, this should not happen.

For comparison, here's one of mine from last Thursday:
[image]

DennisOSRM, 2024-03-26:
Seems like the panels were obstructed some of the time, perhaps? What does the underlying data say?

simonszu, 2024-03-27:
The underlying data looks also quite spiky, but without any gaps (which are showing up in grafana when zoomed in). The same sensor as a dashboard card in Home Assistant looks similar to @dominikandreas but with smaller amplitude, which is caused by my installation form.

closed 2024-04-18 by DennisOSRM