DennisOSRM/hms-mqtt-publisher #122: Release binaries with GitHub Actions

bellackn, 2024-04-17:
it would be great if release binaries were published after a tag/release was created. preferably for multiple architectures. that would make it easier to simply run `hms-mqtt-publish` on a raspberry pi for example, without having to cross-compile on your own or installing the whole rust toolchain on the pi.

DennisOSRM, 2024-04-18:
That's a neat idea. I guess publishing Debian packages could make sense.

bellackn, 2024-04-18:
that's a big step already. just publishing compiled binaries, alongside the source code that's already published on releases in this repo, would be a good first step i think. i suppose publishing debian packages is much more involved.

bellackn, 2025-04-29:
hey, just letting you know that i don't have access to compatible inverters anymore, so i won't do any work on this anymore.