DennisOSRM/hms-mqtt-publisher #123: update_interval is ignored

Matthies, 2024-05-19:
```
 Add-on version: v0.5
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.5.4
 Home Assistant Supervisor: 2024.05.1
```
Although update_interval is set to 60500 (I even tried bigger values like 90000) I can see the data values of the hoymiles device in HA refreshing every 30 seconds.
That alone is not a big problem but I also can see the DTA going offline most of the time in Hoymiles cloud since the add-on is running in HA. Maybe related to the (too) frequent updates in HA.

Matthies, 2024-05-19:
Switching to the nightly build seems to fix this. Not sure if the Cloud offline problem is also solved.

Matthies, 2024-05-19:
Okay, I found issue #52 and #78 but I'm not sure if this is implemented in the HA add-on and if it needs something to activate this cooperative mode.

AFPass, 2024-06-13:
I have the same problem. Update interval every 30 seconds, even though a higher value is set.