DennisOSRM/hms-mqtt-publisher #125: panic: slice index starts at 10 but ends at 0

danilobuerger, 2024-06-16:
Getting 
```
thread 'main' panicked at hms2mqtt/src/inverter.rs:105:61:
slice index starts at 10 but ends at 0
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
After about a minute of running it with a HMS-800W-T2

pimperator, 2024-07-19:
I do have the same issue starting today 12:00

pimperator, 2024-07-19:
```
[2024-07-19T18:41:05Z INFO  hms_mqtt_publish] Publishing to Home Assistant
thread 'main' panicked at hms2mqtt/src/inverter.rs:105:61:
slice index starts at 10 but ends at 0
stack backtrace:
   0: rust_begin_unwind
             at ./rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:645:5
   1: core::panicking::panic_fmt
             at ./rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
   2: core::slice::index::slice_index_order_fail_rt
             at ./rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/slice/index.rs:98:5
   3: core::slice::index::slice_index_order_fail
             at ./rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/slice/index.rs:91:14
   4: <core::ops::range::Range<usize> as core::slice::index::SliceIndex<[T]>>::index
   5: hms2mqtt::inverter::Inverter::update_state
   6: hms_mqtt_publish::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

pimperator, 2024-07-19:
```
thread 'main' panicked at hms2mqtt/src/inverter.rs:105:61:
slice index starts at 10 but ends at 0
stack backtrace:
   0:     0x56552f5ceb2c - std::backtrace_rs::backtrace::libunwind::trace::ha637c64ce894333a
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5
   1:     0x56552f5ceb2c - std::backtrace_rs::backtrace::trace_unsynchronized::h47f62dea28e0c88d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x56552f5ceb2c - std::sys_common::backtrace::_print_fmt::h9eef0abe20ede486
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x56552f5ceb2c - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hed7f999df88cc644
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:44:22
   4:     0x56552f5f8af0 - core::fmt::rt::Argument::fmt::h1539a9308b8d058d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/fmt/rt.rs:142:9
   5:     0x56552f5f8af0 - core::fmt::write::h3a39390d8560d9c9
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/fmt/mod.rs:1120:17
   6:     0x56552f5cbcdf - std::io::Write::write_fmt::h5fc9997dfe05f882
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/io/mod.rs:1762:15
   7:     0x56552f5ce914 - std::sys_common::backtrace::_print::h894006fb5c6f3d45
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:47:5
   8:     0x56552f5ce914 - std::sys_common::backtrace::print::h23a2d212c6fff936
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:34:9
   9:     0x56552f5cff17 - std::panicking::default_hook::{{closure}}::h8a1d2ee00185001a
  10:     0x56552f5cfc7f - std::panicking::default_hook::h6038f2eba384e475
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:292:9
  11:     0x56552f5d0398 - std::panicking::rust_panic_with_hook::h2b5517d590cab22e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:779:13
  12:     0x56552f5d027e - std::panicking::begin_panic_handler::{{closure}}::h233112c06e0ef43e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:657:13
  13:     0x56552f5ceff6 - std::sys_common::backtrace::__rust_end_short_backtrace::h6e893f24d7ebbff8
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:170:18
  14:     0x56552f5cffe2 - rust_begin_unwind
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:645:5
  15:     0x56552f3110d5 - core::panicking::panic_fmt::hbf0e066aabfa482c
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
  16:     0x56552f311732 - core::slice::index::slice_index_order_fail_rt::hd05b814beec6e1b0
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/slice/index.rs:98:5
  17:     0x56552f311732 - core::slice::index::slice_index_order_fail::h0cf4df47d0fb97e4
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/slice/index.rs:91:14
  18:     0x56552f45de68 - <core::ops::range::Range<usize> as core::slice::index::SliceIndex<[T]>>::index::h0c7e067f895d339e
  19:     0x56552f45f273 - hms2mqtt::inverter::Inverter::update_state::ha7e1f8b7c995bd1e
  20:     0x56552f32c10f - hms_mqtt_publish::main::h51c6f1e2031b2f68
  21:     0x56552f326603 - std::sys_common::backtrace::__rust_begin_short_backtrace::ha85f32fa3ab71a59
  22:     0x56552f332749 - std::rt::lang_start::{{closure}}::h9acdb1f36d0bb255
  23:     0x56552f5c5057 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::hd95060ecd5e1ca24
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/ops/function.rs:284:13
  24:     0x56552f5c5057 - std::panicking::try::do_call::h6e8cf51db32a6e4b
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:552:40
  25:     0x56552f5c5057 - std::panicking::try::h3a52eefe24fe3c29
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:516:19
  26:     0x56552f5c5057 - std::panic::catch_unwind::h24c28c23c02c3841
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panic.rs:142:14
  27:     0x56552f5c5057 - std::rt::lang_start_internal::{{closure}}::h705d3c9cbc06ef47
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/rt.rs:148:48
  28:     0x56552f5c5057 - std::panicking::try::do_call::ha21f52ba13158470
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:552:40
  29:     0x56552f5c5057 - std::panicking::try::h5581346bf6aeb1f8
[... cut ...]
  30:     0x56552f5c5057 - std::panic::catch_unwind::h7919645a6b72e25b
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panic.rs:142:14
  31:     0x56552f5c5057 - std::rt::lang_start_internal::h12de51168669836e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/rt.rs:148:20
  32:     0x56552f33273e - std::rt::lang_start::h417bea817f643f9f
  33:     0x7f90ce256d0a - __libc_start_main
  34:     0x56552f3118ba - _start
  35:                0x0 - <unknown>
```

pimperator, 2024-07-20:
update: after the inverter turned off during night it came back properly delivering data
```
[2024-07-19T19:14:13Z INFO  hms2mqtt::inverter] Inverter is Offline
[2024-07-20T03:35:02Z INFO  hms2mqtt::inverter] Inverter is Online
```