DennisOSRM/hms-mqtt-publisher #127: 2x HMS-800W-2T mit einem PC auslesen

tuvok81, 2024-07-17:
Hallo,
da man ja nur einen WR momentan mit der software im selben Netzwerk auslesen kann wäre es schön,
wenn die Software das irgendwie hinbekommt 2 WRs gleichzeitig auszulesen.

theoretisch müsste man das irgendwie umbauen 

[inverter1]
inverter_host = "192.168.178.10"
update_interval = 30500
mqtt_name = "hms800wt2"

[inverter2]
inverter_host = "192.168.178.11"
update_interval = 30500
mqtt_name = "hms800wt2-2"

[simple_mqtt]
host = "192.168.178.3"
username = ""
password = ""

wäre schön, wenn es diese Option geben würde

stefan123t, 2024-07-17:
@tuvok81 the mqtt prefix is hard coded here in simple_mqtt.rs

https://github.com/DennisOSRM/hms-mqtt-publisher/blob/44340dea4de9b0336cb369a7843aac70d3f68bd5/hms2mqtt/src/simple_mqtt.rs#L48-L77

tuvok81, 2024-07-17:
ich bin da nicht so firm drin ich schreibe einfach in der datei wo hms800wt2 steht halt nen -2 dahinter dann sollte das klappen hoff ich :) .. Ich hoffe ja, dass @DennisOSRM  da was aktualisiert :) den Kasten Bier bekommt er, wie er sich gewünscht hat hehe

DennisOSRM, 2024-07-17:
Es ist auf meiner Liste das zu implementieren. Sollte nicht zu viel Arbeit sein, aber evtl komme ich erst in ein oder zwei Wochen dazu.

tuvok81, 2024-07-19:
so ich habe mal einen 2ten ordner gemacht den part abgeändert und compiliert scheint erstmal so zu funktionieren :)

// TODO: this section bears a lot of repetition. Investigate if there's a more idiomatic way to get the same result, perhaps using a macro
        let topic_payload_pairs = [
            ("hms800wt2-2-2/inverter_local_time", inverter_local_time),
            ("hms800wt2-2/pv_current_power", pv_current_power.to_string()),
            ("hms800wt2-2/pv_daily_yield", pv_daily_yield.to_string()),
            ("hms800wt2-2/pv_current_power", pv_current_power.to_string()),
            ("hms800wt2-2/pv_daily_yield", pv_daily_yield.to_string()),
            ("hms800wt2-2/pv_grid_voltage", pv_grid_voltage.to_string()),
            ("hms800wt2-2/pv_grid_freq", pv_grid_freq.to_string()),
            (
                "hms800wt2-2/pv_inv_temperature",
                pv_inv_temperature.to_string(),
            ),
            ("hms800wt2-2/pv_port1_voltage", pv_port1_voltage.to_string()),
            ("hms800wt2-2/pv_port1_curr", pv_port1_curr.to_string()),
            ("hms800wt2-2/pv_port1_power", pv_port1_power.to_string()),
            ("hms800wt2-2/pv_port1_energy", pv_port1_energy.to_string()),
            (
                "hms800wt2-2/pv_port1_daily_yield",
                pv_port1_daily_yield.to_string(),
            ),
            ("hms800wt2-2/pv_port2_voltage", pv_port2_voltage.to_string()),
            ("hms800wt2-2/pv_port2_curr", pv_port2_curr.to_string()),
            ("hms800wt2-2/pv_port2_power", pv_port2_power.to_string()),
            ("hms800wt2-2/pv_port2_energy", pv_port2_energy.to_string()),
            (
                "hms800wt2-2/pv_port2_daily_yield",
                pv_port2_daily_yield.to_string(),
            ),
        ];

tuvok81, 2024-07-19:
OK scheint doch nicht so zu funktionieren :(

[image]

tuvok81, 2024-07-19:
hmpf will doch nicht klappen :( 
("hms800wt2-2-2/inverter_local_time", inverter_local_time), hatte hier ne 2 zu viel aber er liest nix aus :(

[image]

CWempe, 2024-10-27:
## English
Any solution for this without compiling, yet?

I would need this feature, too.

As another workaround I can think of using a second hms-mqtt-publisher and configure it to use a second mqtt broker.
This way you would be able to differenciate between both inverters, even if you cannot change the topic.

## Deutch
Hat schon jemand eine Lösung für das Problem?

Ich brauche auch die Unterstützung für zwei Wechselrichter.
Oder wenigstens die Möglichkeit, das über einen zweiten Docker-Container zu lösen.

Im Notfall fällt mir noch der Workaround ein, dass man einen zweiten MQTT-Broker benutzt, um die Wechselrichter zu unterscheiden.
Also auch einen zweiten hms-mqtt-Publisher, der zwar das gleichen Topic benutzt, aber eben zu einem anderen Broker sendet.

philippselle, 2024-11-08:
Bei mir läuft das Auslesen von zwei Wechselrichter mit folgenden Änderungen:
MQTT Client String angepasst: 
Datei: src/bin/hms-mqtt-publish/rumqttc_wrapper.rs
Zeile: 69

Bei mir waren die ersten 8 Zeilen der Seriennummer anders, als bei meinem 1. WR. Ansonsten musst du noch das Topic ändern lassen:
Datei: hms2mqtt/src/home_assistant.rs
Zeile: 74

Danach einfach den Container local bauen und parallel zum Container des 1. WR laufen lassen.

Dazu gibt es auch einen offenen PR (nicht von mir) hier: https://github.com/DennisOSRM/hms-mqtt-publisher/pull/94/files

DennisOSRM, 2026-10-04:
Danke für die Anfrage! Mit #150 lassen sich mehrere Wechselrichter mit demselben MQTT Broker betreiben: pro Wechselrichter eine Instanz starten und jeder eine eigene `DEVICE_ID` (bzw. `device_id` in der config.toml) geben, z. B. die letzten Ziffern der Seriennummer. Jede Instanz veröffentlicht dann unter eigenen Topics und Home Assistant Entitäten und nutzt eine eigene MQTT Client ID, sodass sie sich nicht mehr gegenseitig trennen. Beim einfachen MQTT ersetzt die Device ID das Präfix `hms800wt2` der Topics. Das nightly Docker Image und das nightly Add-on haben es bereits, die stabilen Versionen folgen mit dem nächsten Release.

closed 2026-10-04 by DennisOSRM