DennisOSRM/hms-mqtt-publisher #138: toml config unparsable: Error

ssp164, 2025-05-12:
When I start the Docker container I get the following error and the container terminates.
```
2025/05/12 06:29:43,stdout,note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

2025/05/12 06:29:43,stdout,"toml config unparsable: Error { inner: Error { inner: TomlError { message: \"invalid string\nexpected `\\"`, `'`\", original: Some(\"inverter_host = \\"\\" \n\n[home_assistant] \nhost = \\"\\"\nusername = \\"\\"\npassword = \\"\\"\nport = \n\n\"), keys: [], span: Some(84..85) } } }
"
2025/05/12 06:29:43,stdout,thread 'main' panicked at src/main.rs:55:52:

2025/05/12 06:29:43,stdout,[90m[[0m2025-05-12T04:29:43Z [32mINFO [0m hms_mqtt_publish[90m][0m loading configuration from /config.toml

2025/05/12 06:29:43,stdout,[90m[[0m2025-05-12T04:29:43Z [32mINFO [0m hms_mqtt_publish[90m][0m Running revision: b3759dadd318c48b8f84b04a2b94ff0c230008f1
```

DennisOSRM, 2025-05-14:
Would need the config file (sans password) to debug. The message says it contains errors.

Rouji, 2025-06-19:
The image on dockerhub seems to be somehow different than current master, dunno what's wrong there.  
*But* it works fine if you build it yourself. But only if you specify *all* variables, including `UPDATE_INTERVAL`.   
See https://github.com/DennisOSRM/hms-mqtt-publisher/blob/main/docker/entrypoint.sh

toml parser probably unhappy if you have a `some_variable = ` line in there or something

kaischmitz, 2025-06-27:
How is the interaction between this config file and the settings from environment variables supposed to work?